MaslowCNC/Electronics #7: Check for continuity on 12V

blurfl, 2017-02-03:
In the area above and to the right of U$18 it looks like the top side copper pour (12V, I think) might be broken by the thru-hole of INCD1 and the keep-out around the ground pad of U$18. Is this the main 12V path to the rest of the board?
It might just be a difference in the way my tools are rendering the pours, though...
[image]
[image]

BarbourSmith, 2017-02-04:
That is definitely a real continuity issue. Thanks for catching that. Definitely the low point of my morning. I broke continuity there wiring up the 4th axis. I was in a huge rush to get those out because they are our longest lead time item. I am going to do a re-design next week and move all the components down so that top area is less cramped. For now I've added a wire to patch things in #8 just so the files up there aren't broken. 

I checked in with the factory and its too late to do anything to the boards which are in production so I'll have to solder a wire to each of those which is pretty far from ideal. I guess this is why we do the whole beta thing.

A huge thank you again for catching that. That's a pretty big issue.

BarbourSmith, 2017-02-04:
I'd like to give you credit for having spotted the problem in next week's update. Is that OK? If so, do you have a way you would be preferred to be identified? (real name, blurfl, anonymous community member...etc)

blurfl, 2017-02-04:
I’d as soon not be identified, if that’s OK. Participation is important, glad I could help.

While I’ve got your ear, I’ve fished an old Arduino Mega 2650 out and put the firmware on it, but it seems pretty unhappy talking to GroundControl. Is it reasonable for me to try anything without the power board? Should I open a topic on the forum, or hold off until boards get to the betas?

Keep up the great work, you and Hannah are running an exciting project in an exemplary manner!

blurfl, 2017-02-04:
I’ve got stacks of boards with similar errors from previous projects.  A beta wouldn’t be proper without at least one “green wire”. :) The SMD Hbridges make good sense from a manufacturing angle, though. Worth the effort of the change.

BarbourSmith, 2017-02-04:
I'll order green wire to make the joke complete :-)

Thanks for the support. We do our best, but we really couldn't do it without you.

I wouldn't expect the lack of a power board to matter to the serial connection. What types of issues are you seeing? I tried to increase the baud rate last week and started running into issues and stepped it back so making sure you are using the current version might make a difference. I'm working on getting a Mac computer and a Raspberry Pi set up as two more test platforms in the next week. I was seeing some connection issues on the Mac which I haven't explored yet. What type of platform are you running GC on?

It might be worth filing a bug on GC for this or making a thread in the forums. I think it's something important to explore.

closed 2017-02-05 by blurfl