MaslowCNC/Electronics #12: Power LED should show 12V power instead of 5V

blurfl, 2017-03-26:
There presence of 5V power can be assumed if the USB connection is made; the presence of 12V power is necessary for the motors, and using this led for that would make it easy for a user to know he has the power supply correct. Relabel it '12V' or 'Motor Power'.

BarbourSmith, 2017-03-27:
Glad there is an issue for this, I agree 100%

blurfl, 2017-03-27:
It should be an easy edit, but if the boards have already been ordered…

BarbourSmith, 2017-03-27:
😬 It's hard to change things once the order goes out.

I've been putting off placing the next batch order to get as much feedback as possible rolled into them, but I think I need to ship them in the next few days. The basic design will be the same as in PR #9 with a 12v LED indicator added.

blurfl, 2017-03-27:
That should do it 👍

closed 2017-04-11 by blurfl