MaslowCNC/Electronics #13: Power supply connection

neilfrazerm, 2017-03-31:
Instructions  at setep for for those internationals without a power supply provided.
Please advise / insert the size of the connector and/or suggest supplying one in the kit if no power supply provided.
Thanks
Neil

BarbourSmith, 2017-04-01:
I checked in with Hannah and I realize this is totally my fault. I intended for those of you who live in countries with different power requirements to still receive the power supply, but just no power cord. 

I chose a power supply which works from 100-250vac 50/60hz, and which uses a standard computer power cord so that they would be easy to find with the correct plug for your country.
[image]

I didn't make that clear enough and we ended up sending you no power supply at all. The power supply is on the way, but I understand if you don't want to wait for shipping time and want to buy one locally. I'm really sorry for the mix-up, it was completely my fault.

Fortunately, the other side of the power supply is also the most common connector I could find. Its a 5.5mm barrel connector with a 2.1mm pin. A 5.5mm barrel connector with a 2.5mm pin will also fit. The center pin is positive (standard). The supplied power-supply is a 5A  12V dc supply with self resetting fuses for over-current and over voltage protection, but almost any 12volt supply will work. 

[image]

The power supply for the motors is isolated from the electronics so the quality of the voltage is not critical. Almost any 12 volt will work. I am more than happy to look at particular supplies if you want, and if you can find one locally that will ship more quickly than the original one which is in the mail, we are more than happy to pay for it and have it sent to you.

Again, I'm sorry for the mixup.

neilfrazerm, 2017-04-01:
Thanks awfully for sending out the supply, as you have already done so, and I am away for the next week, I shall wait for yours...

I had found the right connector from Maplins (UK Farnell equivalent) .

I do understand the pressure you are under and really do wish you rather well old chap.

Sincerest best regards,

Neil

(London)

[cord]<https://cloud.githubusercontent.com/assets/9359447/24580917/0617c318-16c6-11e7-83b0-c81c48bd911a.JPG>

[https://cloud.githubusercontent.com/assets/9359447/24580917/0617c318-16c6-11e7-83b0-c81c48bd911a.JPG]

[plug]<https://cloud.githubusercontent.com/assets/9359447/24580994/d7c714b2-16c7-11e7-86e3-d7d662589767.JPG>

[https://cloud.githubusercontent.com/assets/9359447/24580994/d7c714b2-16c7-11e7-86e3-d7d662589767.JPG]

blurfl, 2018-01-10:
@BarbourSmith, this looks like it could be closed, but your reply and its pictures look like good wiki fodder 😀

closed 2018-01-10 by BarbourSmith