MaslowCNC/Electronics #14: Motor Control Pins - are two GPIO pins necessary?

blurfl, 2017-05-22:
It looks like the two motor control pins are always used in complementary states in Motor.cpp. Could each motor be controlled by one GPIO and an inverter to use fewer GPIO pins?

BarbourSmith, 2017-05-23:
No, we could really use just one pin and an inverter. 

Using two pins keeps the part count down (we've already got the pins).

In theory we could do active braking of the motor by using both pins high (or maybe it's low), but I haven't played with that option yet.

blurfl, 2017-05-23:
Using one pin eases the task of adapting other cpu options (ESP32). The inverters could live on the adapter board instead of on the motor board. 
With the present gearing setup, would adaptive braking accomplish much?

closed 2017-05-23 by blurfl