MaslowCNC/Electronics #18: Missing Heatsink from BOM

JDogHerman, 2017-06-30:
In the pictures I see a heatsink shown but it is not an item on the electronics BOM. Can this be added?

BarbourSmith, 2017-06-30:
Great point! Yes it can

BarbourSmith, 2017-06-30:
I've added it to the BOM. Its good that it is in there to remind people to order one when buying parts. I can't give you a digikey number for the part because we had ours custom made, but it's not a critical part. Digikey sells a number in the right size range.

JDogHerman, 2017-06-30:
Will there be a method of purchasing these "one off custom" parts from the MaslowCNC org?

JDogHerman, 2017-06-30:
What sort of thermal resistance is required?

JDogHerman, 2017-06-30:
Would a part like Digikey: 345-1103-ND work? Its a "little" larger but it already has the adhesive base.

BarbourSmith, 2017-07-01:
That one looks perfect. I measured the space and it will fit. I would recommend the taller 25AB model because the short one might not move enough heat.

If you want to throw it in the BOM that would be fantastic. 

We're probably never going to sell the heatsinks individually because it's not really cost effective, but we will probably sell the whole arduino shield as a board with the heat sinks in the next few weeks

JDogHerman, 2017-07-01:
Added the suggested heatsink to pull request #19

schlaubi0666, 2017-07-12:
the sugested one is also 6.4mm in heigth, but the one on the Picture is much taller ...
is this ok , to use the not so tall Version?
or better use ATS1319-ND ?

JDogHerman, 2017-07-12:
@schlaubi0666 It looks like the ATS1319-ND does not have numbers for thermal resistance while being naturally cooled.  Its also more than 3 times the price. (might be necessary) @BarbourSmith does this part @schlaubi0666 suggested look like a more fitting part?

JDogHerman, 2017-07-12:
The nice thing is since the H-Bridge is over-temped protected and it is a closed loop system, if the H-Bridge does go into over-temp the encoder will detect that it has missed steps and the cut wont be lost.

In a future board design it might be nice to use L298N as it has much better thermal transfer numbers vs the L298P. 3C/W vs 13C/W*while using a aluminum substrate (not used in current boards)

BarbourSmith, 2017-07-12:
While they say it's over temp protected, we've blown up enough at this point to put that in the false marketing category ☹️ 

I think I tracked down a part that I was looking at as a replacement in the future, but I can't remember what it is off the top of my head. Moving away from the L298 family is in the future.

blurfl, 2018-01-10:
@BarbourSmith, have you got a heatsink (and thermal tape) part number for the BOM? 

Also your comment "Moving away from the L298 family is in the future." [Issue #23](https://github.com/MaslowCNC/Electronics/issues/23) is a discussion about this general topic - any further thoughts?

BarbourSmith, 2018-01-10:
We ended up having them custom made 🙄 so I don't have a part number. It's weird how it's cheaper to have a heatsink made to spec than to buy one off the shelf

I hadn't given too much thought to moving away from the L298 family because we stopped blowing them up with such regularity...I still think moving is a good idea...I am a little worried about possible effects to the PID tuning

blurfl, 2018-01-10:
I think I've ordered replacements for these.I think I've got Digikey part numbersfor the [thermal tape](https://www.digikey.com/product-detail/en/3m-tc/15MM-15MM-10-8810/3M10310-ND/2649855) and the heat sinks, but I haven't found the heat sink one yet 🙃. I'll keep looking and reply to this issue.

blurfl, 2018-01-10:
I was wrong, the heat sinks were for a different chip family, not applicable. Are there enough on hand to offer them in the Maslow store?

blurfl, 2018-01-12:
[This](https://www.mouser.com/productdetail/aavid-thermalloy/374124b00035g?qs=sGAEpiMZZMttgyDkZ5WiulSjpO3gDOKQ4Eu1ZE9AH98%3D) heatsink from Mouser is 23mm x 23mm and 18mm high, so slightly shorter but in the ballpark. In stock, $2.35 each, includes the thermal tape.

[image]

BarbourSmith, 2018-01-12:
That looks like a perfect solution!

blurfl, 2018-01-12:
Shall I add it to the BOM in the same PR that updates the schematic and silkscreen, or would you like them separated?

BarbourSmith, 2018-01-12:
Having a lot of pull requests that change the board files themselves can be tricky, but doing this as a separate PR should be safe I think

blurfl, 2018-01-12:
Oops, I'll pull the PR back and separate them - sorry, jumped the gun!

blurfl, 2018-01-12:
While I'm at it, should I be saving these files in a particular version of EAGLE v8.5 is current, but it offers an option to save for EAGLE v7.x ?
Also, I'd like to remove the backup files and lock file that EAGLE leaves around - only the .sch and .brd files in the repo.

blurfl, 2018-01-12:
@BarbourSmith, here's the situation - The pin label changes affect the schematic visibly and the board invisibly, so both must be changed together. The silkscreen change affects the board but not the schematic, but both files should still be submitted together.
 I did the schematic change first as it made a bigger change. Shall I remove the diode symbols from the board file that goes with the schematic change, open a PR for just the schematic change, then add the diode symbols to the silkscreen after the schematic PR is merged?

BarbourSmith, 2018-01-12:
Right, I see the issue

Is it possible to submit just both changes at once? I think that would be totally OK

blurfl, 2018-01-13:
This was addressed in PR#41 - is it Ok to close this?

BarbourSmith, 2018-01-13:
I vote to close this one as well. You are really knocking these off!

closed 2018-01-13 by blurfl