MaslowCNC/Electronics #20: Heatsink substrate

JDogHerman, 2017-06-30:
I might be wrong in this, but would it be good to add a thermal substrate copper pad for the H-Bridge ICs to sit upon?

BarbourSmith, 2017-07-01:
I think it's a great idea. We actually had one in an early version of the PCB, but I took it out because our PCBA people were having enough trouble and I wanted to make things really really really easy for them. It might be time to put that back in. The ground plane runs right under each chip to make doing exactly that easy.

JDogHerman, 2017-07-01:
Sorry I opened this and then looked closer at the traces and saw that the traces run underneath the ICs.

dmouw25, 2017-09-01:
Is it possible to Reopen this?   These packages are designed to dissipate most of the heat through slug on the bottom of the package.  If you notice, the datasheet does not give thermal values for junction to case.   You would likely get better thermal performance with a redesigned PCB and no heatsink.

JDogHerman, 2017-09-01:
Reopened as a feature request.

blurfl, 2018-01-10:
I believe this was addressed in the v1.2 layout. @JDogHerman, @dmouw25 does that look correct to you? If so, we could close this.

BarbourSmith, 2018-01-10:
I second that it was a great suggestion and has been added

closed 2018-01-10 by blurfl