MaslowCNC/Electronics #21: how to get a pcb

schlaubi0666, 2017-07-12:
i missed Kickstarter, and you do not know it you are selling kits again ;-(

1. do you will provide an partlist soon ?

the most difficult part to get is the pcb of the Driver ....

2. you have some spare modules to sell ?

3. it not, can you provide the gerber data to order my own pcb ?
4. what is the spec/Parameter of your ordered pcb's ?

br

Dirk

BarbourSmith, 2017-07-12:
We're hoping to start selling individual parts really soon, so hopefully in a couple weeks you will be able to buy an arduino shield from us in the next few weeks.

Until then, we do have a parts list here: https://github.com/MaslowCNC/Electronics/blob/master/BOM.txt and the gerber files are available here: https://github.com/MaslowCNC/Electronics/tree/master/PowerDistribution if you want to make your own.

The specs for the boards are nothing special. Maybe 16mil trace width and 1oz copper. Anyone who can make PCBs should be able to make one for you.

JDogHerman, 2017-07-14:
@schlaubi0666 I am in the same boat with you. I put in an order for a PCB and got more than I needed (min quantity). PM me if you would like me to get you one at cost+shipping.

schlaubi0666, 2017-07-15:
the part list  a was taking about , is the whole list , not just the electronics ...

schlaubi0666, 2017-07-15:
@JDogHerman : i did not found an way th contact you (PM) , so yes i would like to have on. plz. write me an email to my username + @gmx.de

JDogHerman, 2017-07-17:
@schlaubi0666 I got an email bounceback from the gmx.de server. Is there another address I can contact you at?

ndefoor, 2017-07-18:
@JDogHerman Could you message me  nathandefoor@hotmail.com I would like to see about getting one of those extra boards. Thanks

mgood83, 2017-07-20:
@JDogHerman could you also message me in regards to purchasing a board from you. I was just about to get one made, but if you already have spare, even better. Happy to pay send payment for cost + time +shipping. Thanks. mwgoodlett@gmail.com

clintloggins, 2017-07-20:
@JDogHerman please count me in if you have another pcb. More than happy to pay a premium!! Also, any of the changes that you discussed I am willing to test and give feedback. Thanks!
clintloggins@gmail.com

schlaubi666, 2017-07-20:
thanks for selling the bord now in your shop , i will grab one there ....

ndefoor, 2017-07-26:
I am looking for the specifics on the motors.  Does it matter if the rpms are slightly different?

JDogHerman, 2017-07-26:
@ndefoor There is a discussion on alternative motors options on the forums. 

[forum link](http://forums.maslowcnc.com/t/for-those-that-were-not-part-of-the-kickstarter-where-to-obtain-the-main-motor-worm-gear-encoder/120)

JDogHerman, 2017-07-26:
This issue can be closed.

closed 2017-07-26 by BarbourSmith