MaslowCNC/Electronics #22: Silkscreen is missing LED polarity

JDogHerman, 2017-07-14:
It might be nice to see the LED polarity shown on the silkscreen for those of us that are sadistic enough to build our own boards.

Also why not show a website address? maslow.cnc :+1

BarbourSmith, 2017-07-17:
Those are both great suggestions

blurfl, 2018-01-10:
I'm going to be into the design files working on the schematic for [issue #36(https://github.com/MaslowCNC/Electronics/issues/36). I'll open a separate PR to add this and update the board .jpg as well.

blurfl, 2018-01-11:
Rather than draw an LED symbol, I've put a plus sign opposite the positive end of each of the two LEDs:
[image]

If that's satisfactory, I'll put in a PR with the change.

BarbourSmith, 2018-01-11:
I think that looks great!

BarbourSmith, 2018-01-11:
There is a small dot on the LED part, does that not indicate polarity?

[image]

[image]

BarbourSmith, 2018-01-11:
My read of the schematic is that the polarity should be like this:

[image]

[image]

blurfl, 2018-01-11:
Thanks Bar, you're right. Hard to turn my thinking around to 12V power plane and routed ground. The dot _does_ indicate polarity, but I think it wouldn't appear on the silkscreen of a PCB in a maker's hand because it's in a keep-out area. The plus would address that. You've got a good point, though - SMD LED's use a dot to mark the cathode end. How best to address the issue, dot-at-cathode or plus-at-anode? @JDogHerman, what's your opinion?

BarbourSmith, 2018-01-11:
I think the plus is a great idea! I was just pointing out the dot because I was curious what it meant. You are absolutely right that it does not show up anywhere on the PCB. I actually had an email from the manufacturer this morning to confirm the diode directions so even they are confused 😛 

The plus will be much better.

blurfl, 2018-01-11:
Just to stir the pot, how about:
[image]

and
[image]

BarbourSmith, 2018-01-11:
SLICK! I love it

blurfl, 2018-01-13:
This was addressed in PR#42 - is it Ok to close this?

BarbourSmith, 2018-01-13:
Also beautifully taken care of. I vote to close.

closed 2018-01-14 by blurfl