MaslowCNC/Electronics #23: thru-hole Passive components

JDogHerman, 2017-07-14:
Since this shield has lots of room on it what if there was thru-hole passive components holes added to the board in addition to smb. That way the board could be built as a kit easier for mortals.

BarbourSmith, 2017-07-17:
It might seem like there is a lot of space, but a lot of the open copper is actually left intentionally open because we have a lot of current flowing across the board. I'm open to the idea if someone can design one that looks good, but I think we'd be better off doing a second board that's the through hole version if we want that

JDogHerman, 2017-07-17:
If the XH connector housing was moved down it would allow for more room for the A+ B+ motor lines to be widened and permit though hole diodes to be added. (in addition to reducing the trace length)

The current limiting resistor and LEDs have plenty of board room and the filter caps could be moved closer to the supply rather than near the H-Bridges. #27 

In addition if the MP3 H-Bridge was moved to the right it would help provide less thermal coupling with the other H-Bridge.

BarbourSmith, 2017-07-17:
Those are great points! If you make those changes, I'd be more than happy to merge them. My only reservation is that I think I'm going to redesign with a new H-bridge chip so we're probably going to see a lot of changes soon

JDogHerman, 2017-07-17:
@BarbourSmith what H-bridge chip are you thinking? Maybe I can help propose a layout with the new chip in mind.

BarbourSmith, 2017-07-17:
That would be great! I'm thinking about using the L6206 (https://www.digikey.com/product-detail/en/stmicroelectronics/L6206PD013TR/497-3643-1-ND/677626)

JDogHerman, 2017-07-17:
Interesting. The L6206 has integrated diodes, and higher current capacities, but if thermal considerations havn't been built into the PCB. The heat will skyrocket with 62 C/W! 🔥

BarbourSmith, 2017-07-17:
Good spot, that's something we need to consider. 62C/W is nuts, how does anyone keep that cool?

JDogHerman, 2017-07-17:
Looks like you can get it down to 15C/W by having it 

> Mounted on a multilayer FR4 PCB with a dissipating copper surface on the top side of 6 cm2 (with a thickness of 35 µm),
> 16 via holes and a ground layer.

then maybe adding heatsinks and active cooling could keep it cool. ❄️ 

Also did you see that digikey has limited numbers of this pn?

BarbourSmith, 2017-07-17:
I (clearly) didn't do too much research into finding a replacement part yet, and I'm rapidly becoming convinced that the L6206 is not the best choice. Are there any other candidates out there that look more promising?

JDogHerman, 2017-07-17:
What features are you hoping to achieve? 
More current? Higher voltage? Better thermal characteristics? Faster switching?

BarbourSmith, 2017-07-17:
I'd say the #1 thing we need is durability. I'd like a chip that we can short without blowing it up, and which doesn't overheat easily. Other than that our needs are for a pretty basic 12v 2A+ H-Bridge

JDogHerman, 2017-07-18:
Have you seen this? [TLE5206-2GCT-ND](https://www.digikey.com/products/en?keywords=TLE5206-2GCT-ND)

Low Thermal Resistance, Internal Inductive Diodes, Short circuit protection with a error pin so software could alert the user that there was an issue. Cost is higher but isnt bad. 

Better pcb thermal design features and a heat sink would still be necessary but the chip should have better reliability since it will have more headroom vs the existing [chip 497-3624-1-ND](https://www.digikey.com/product-detail/en/stmicroelectronics/L298P013TR/497-3624-1-ND/677613)

BarbourSmith, 2017-07-18:
Crikey, I love it! Excellent find! The price is pretty reasonable even with one needed per motor. The board will still be a couple $ more, but for full output short circuit protection it's absolutely worth it.

I'll see if I can throw together a board to test them this weekend, when I have PCBs made, I'd be more than happy to send you some to test too if you'd be interested.

blurfl, 2017-07-18:
I’d like to test or help with the design too, if I might. 
When laying it out, here are some things to think about:

- Could we re-think which pins are used for board ID? The present approach interferes with the ICSP pins, making porting to other processors harder and making it harder to use SPI. Too, using a pull-up/down resistor adds parts count, but protects the gpio pins.
- As we don’t need the braking function, the new layout could use hardware inverters to handle the complement motor drive signals.
- It would be harder to deal with heat sinks, but putting the motor drivers on daughter boards or using the thru-hole version of these chips would make it easier for mortals to repair. Selling motor drive daughterboards as spares would mean less electronic waste when the drivers need replacement.

JDogHerman, 2017-07-18:
@BarbourSmith that sounds great. If you want help with ideas on the design of the PCB let me know. I would be thrilled to test or help design. Just let me know how I can help. We could even add this as a project to github.

@blurfl I too would think a new method for identifying board revisions might be nice.  In fact if there was interest in changing the H-Bridge communication path to SPI there are other chip that could fill that need.

blurfl, 2017-07-18:
Having SPI available would make it possible to add SD card support. 

- gcode transfer serial buffering is an ongoing issue. With a local file system we could transfer separately from the cutting process and the read from there while cutting instead. 
- This could make off-line operation possible as well. 
- Finally, some other processor boards have higher speed, but don’t have the non-volatile storage of the Mega. An SD card would open up those possibilities as well.

 So, while I build my wish list, let me add at least the traces for a micro SD card :)

BarbourSmith, 2017-07-19:
Any and all help is more than welcome, if you want to have a go at putting together a board with the TLE206 or any other chip that you think is worth a go I would love to try it out!

I agree that adding an SPI interface could be nice. My general preference is to keep the whole machine as simple and low cost as possible to keep it reliable and affordable, but that doesn't mean we couldn't have the pins broken out to attach an SD card daughter board. Going to SPI for the motor controllers is also an option. I used the simplest most reliable interface for the boards so far because I wanted to focus on getting as simple and robust a system as possible built, but now that we have a stable platform to work from, there's no reason not to try new things.

JDogHerman, 2017-07-19:
Here is an SPI H-Bridge I was also looking at. https://www.digikey.com/product-detail/en/infineon-technologies/TLE6209RAUMA2/TLE6209RCT-ND/2231220

blurfl, 2017-07-19:
That’a an interesting chip - bigger and more expensive, but it makes the detailed failure info available.

Have you seen anything that could take the burden of interfacing with the motor encoders? That’s presently handled using interrupts on the Mega, and the code for that is inline assembly in order to keep up. I think it might be limiting how fast we can travel. Surely something else could watch the encoder pulses and provide direction and step count when queried. I’ve seen this [LSI-LS7184](http://anaheimautomation.com/manuals/ics/L010546%20-%20LSI-LS7183,%20LSI-LS7184.pdf), but they aren’t widely available.

blurfl, 2017-07-19:
I see there is an SPI version, LS7366, that includes the counter as well, but only available from the manufacturer :(

blurfl, 2017-07-22:
@JDogHerman, it doesn't look like the  TLE5206 will do speed control. Am I missing something? I'm disappointed, I like the TO220 thru-hole option.

mrfugu, 2017-08-15:
If you're considering a significant re-design of the Shield board, please consider adding the necessary provisions to allow installation of a cheap (iOS compatible) [Bluetooth Module](https://www.itead.cc/serial-port-ble-module-master-slave-hm-10.html) .

dmouw25, 2017-08-29:
Is there any description / datasheet for the motors used?  I can't find it in the wiki.  Thinking about starting a redesign of the board with better heatsinking of the ICs, and better diodes, but want more details on the motors before I start the design phase.  Board would be optimized for automated assembly, not hand assembly.

blurfl, 2017-08-29:
A description from the [wiki](https://github.com/MaslowCNC/Mechanics/wiki/DIY-Kit-Mechanical-Equivalent-Parts):
Motor/Gearbox/Encoder	(291:1 reduction gearbox with worm gear to prevent back driving motor, 30kg/cm torque, 20 rpm, 12 volts)

Some discussion on the forum, [here.](http://forums.maslowcnc.com/t/for-those-that-were-not-part-of-the-kickstarter-where-to-obtain-the-main-motor-worm-gear-encoder/120/35)

mrfugu, 2017-08-29:
@dmouw25 In tinkering with the shield, attempting to get a bluetooth module integrated, I'll note that it would be very useful if all the unused pins, but most especially the digital pins, passed (unmolested) through the shield and were able to be accessed from 'above'.

JDogHerman, 2017-09-10:
I created a branch design using the TLE5206-2G not sure how people feel about it but here it is. 
https://github.com/jdogherman/electronics/tree/TLE5206-2G-Board

blurfl, 2017-09-10:
This looks good! I wonder if a few more vias non the M1-, M2- and M3- lines would be advisable. Not much current capacity in a via! Alternately, make those larger to allow a wire thru-soldered.

JDogHerman, 2017-09-10:
@blurfl good point! I have updated the design with 4 vias per leg.

blurfl, 2017-09-11:
I'm not seeing the 4-via version in your repository yet?
Are all the signals the same, only the H-bridges changed?

JDogHerman, 2017-09-11:
wow somehow I missed adding the new file. It's there now. The signals are all the same but the 12v lines have more vias.

blurfl, 2017-09-11:
I hate to say, but the commit show 0 changes in 0 changed files. GitHub is a fickle mistress...

JDogHerman, 2017-09-11:
Sorry been under the weather and I must not be thinking too clearly. I see the changes on https://github.com/JDogHerman/Electronics/commit/d9fb86bcf717d573a443468bce3ea259ec66f5bf Can you check that you can see the changes?

blurfl, 2017-10-19:
@JDogHerman,  I've been looking at the TLE5205 and TLE5206 again lately. Have you looked at the firmware changes to drive these?

blurfl, 2018-01-29:
I have a working board using the TLE5206. It requires changes to System.cpp and Motor.cpp for new pins and a different method of controlling the chip,

blurfl, 2018-06-01:
With the release of the TLE5206 board in the Community Garden, most of the ideas have been implemented with the exception of the connection for a Bluetooth module. 
 Would it make sense to close this?

closed 2018-06-01 by JDogHerman