MaslowCNC/Electronics #27: Capacitor location & are C3 & C4 needed?

JDogHerman, 2017-07-17:
I'm looking at the board and I was wondering if C3 & C4 were really necessary. Since C1 & C3 (5v filter) are in parallel and C2 & c4 (12v filter) are the same, would it not make sense to just use a larger capacitor and move those caps away from the warmer and towards the supply?

This would allow better heat dissipation to the H-Bridge along with reducing the part count.

BarbourSmith, 2017-07-17:
That's a great point. We could probably get away with just one set of filtering capacitors, especially since the only parts pulling from that 5v rail are the two H-bridges and an LED which won't add spikes

dmouw25, 2017-08-27:
I would highly recommend leaving both Capacitors in place.  Actually the C1 & C3 should be moved closer to the IC to prevent power glitches from causing odd behavior.    Adding a bulk capacitor to the VDD_12V would probably be a good idea as well since the motors are drawing significant amounts of current.  Application diagram in datasheet for the motor driver had a 470uF electrolytic.   

Package size on the ceramic capacitors could be reduced from 0805 to 0603 to make layout easier, but 0805 is better for ESD (ESD can damage 0603 capacitors, but 0805 can generally absorb the energy without damage).  If this was my design I would place an ESD rated 0805 capacitor at the power connector and 0402 at the ICs for local decoupling.

blurfl, 2017-08-29:
I agree, leave the caps. They are recommended by the chip manufacturer.

closed 2017-08-29 by JDogHerman