MaslowCNC/Electronics #31: J4 - Power Jack Part Number

blurfl, 2017-10-29:
Is it possible that the part used for J4, the 12V power jack changed from the device shown in the BOM?
The BOM shows a CP-002A-ND, which has a 2mm center pin. The jack on the v1.1 board has a 2.5mm center pin. The DigiKey part number for that would be CP-002B-ND.

BarbourSmith, 2017-10-30:
Great catch @blurfl 

I checked the invoice and the part we are using is [CP-202BH-ND](https://www.digikey.com/product-detail/en/cui-inc/PJ-202BH/CP-202BH-ND/408451 )

I will update the BOM right now

blurfl, 2017-10-31:
I see, that one has the higher (5A) current rating. From a company from Tualatin, OR., too 😄

BarbourSmith, 2017-10-31:
No way! That's a 15 minute drive away and I had no idea. Maybe we'll drop by to say hello

closed 2018-01-10 by blurfl