MaslowCNC/Electronics #39: Add ability to drive stepper

davidelang, 2018-01-05:
see https://github.com/MaslowCNC/Firmware/issues/355

since we are have four channels available, it would be good to add an extra header to the board that would let us use two channels together to drive a stepper (ideally using the same pinout that is common for 3d printers)

combined with a software way to drive this (even inefficiently), this would let people use off-the-shelf stepper driven Z axis kits/design with the maslow.

blurfl, 2018-01-10:
This would involve a fairly major re-design of the board.

BarbourSmith, 2018-01-10:
Yes, it would probably be easier to make a new board. I am going to add the tag wontfix because the suggestion is a good one, but we are not actively working towards addressing it

davidelang, 2018-01-10:
On Wed, 10 Jan 2018, Scott Smith wrote:

> This would involve a fairly major re-design of the board.

I don't think it would.

a stepper driver is two H bridges, one for each set of coils. The pre-built 
steppers add extra logic to simplify management of the two h-bridges so that you 
can have one pin for direction and another for step and the chip takes care of 
everything else.

But if we just were to wire the two H-bridge halves to one plug, the rest of the 
logic could be done in software.

It's not efficient, and not pretty, but I can't think of why it wouldn't work.

Remember, we are not needing microstepping here, the stepper (@200 steps/rev) is 
being connected to a lead screw (@8 steps/in or more), resulting in one step 
being ~ 0.00125". That should be more than sufficient for the Z axis. I'd expect 
that we could get by with a resulution 1/10 that (0.0125") without a problem.

So a simple state table in software that we step through to toggle the different 
combinations of power through the h-bridges should be easy to do.

Doing something to drive these at less than full power (pwm) shoudn't be that 
hard either.

blurfl, 2018-01-10:
I'd call it a major re-design as it involves adding another connector and traces for the connector and the presently unused half of the second H-bridge. Further, the present design has spread the chips out for thermal efficiency, leaving no room for a fourth connector.

blurfl, 2018-01-10:
If one wants to drive a stepper, look into driving a stepper driver board using two of the AUX pins. Logic power, ground, step and direction are enough for the Maslow side and are all available from the AUX pins.

BarbourSmith, 2018-01-10:
Great suggestion @blurfl I think driving a stepper off of the AUX ports makes so much more sense and would be a much cleaner build. Ok to close this as a hardware issue and make it a software issue if we decide to go that route?

ProtoMakerOpenTech, 2018-09-10:
Currently working on a maslow variant for the open source market. Going to work on keeping it as backward compatible as possible and will be building a new board for stepper control, support for a lathe axis and VFD. Anyone interested in contributing let me know. Most of my machine design experience is in the 3D printing space and came across the maslow as a possible tool for prototyping.