MaslowCNC/Electronics #45: LED Current too high

dmouw25, 2018-01-13:
The current through the 12VPOWER LED is too high and power dissipation of R2 is very high.   LED is rated for 25mA maximum and LED is currently being driven at 29mA ((12V-2.2V)/330 ohm).   Resistor R2 is currently dissipating 290mW typical.  Increasing resistor R2 to 1000 ohm would allow a normal 0805 resistor to be used instead of thick metal film resistor selected for a small cost savings and will reduced board power dissipation by 190mW.   This will set LED to a  nominal current of 10mA and worst case current at ~12mA.   This will match the 5V LED current,   If maximum brightness is desired, R2 can be set to 510 ohm, which should guarentee that LED current stays below 25mA (assuming 12V supply stays under 13V)

blurfl, 2018-01-13:
A good catch! Great that you're checking deeply into the circuitry. Do you want to do a PR on this, or shall I?

dmouw25, 2018-01-13:
Not sure how to do a PR, so could you do it?

blurfl, 2018-01-13:
I'll be glad to. To summarize, I'll change the value of R2 to 1K on the schematic and the BOM, and look up a 1K SMD thin-film resistor for the BOM, yes?

dmouw25, 2018-01-13:
Correct.   I would personally go with 541-4162-1-ND at digikey But that is mostly because that is a company we use at work.

blurfl, 2018-01-13:
I'm a great fan of DigiKey too. Thanks for the part number 💯

BarbourSmith, 2018-01-13:
Excellent suggestion! Let's change the part in the BOM and change the value for that resistor in the schematic. 👍 👍 💯

blurfl, 2018-01-18:
Ok to close this?

BarbourSmith, 2018-01-18:
I vote to close. Well added!

closed 2018-01-18 by BarbourSmith