MaslowCNC/Electronics #46: Catch Diodes are not fast recovery diodes

dmouw25, 2018-01-13:
The 4004 diode is a general purpose diode and is not designed as a fast recovery diode.  Datasheet for motor driver recommends "The external bridge of diodes D1 to D4 is made by four fast recovery elements (trr ≤ 200 nsec)".   A quick search of digikey found the ES1JFSCT-ND which has a reverse recovery of 35nS vs the 4004 which is not specified.  I suspect we should really go with a schottky diode such as the SSA210 (digikey SSA210CT-ND), but my expertise is  more in DC/DC power supplies and design for automotive EMC.   I will ask around at work of a couple of graybeards and see if they have any advice on an appropriate diode for this application

BarbourSmith, 2018-01-13:
Excellent point! Let's change this.

blurfl, 2018-01-13:
Let's wait to hear what the 'graybeards' have to say. When we hear back, we'll "make it so" 😁

dmouw25, 2018-01-17:
So I talked with the guys at work and going to the schottky is a good idea but with one caveat that requires testing.  That said I am not convinced it is worth the cost unless it actually helps performance.  

Caveat:
Schottky diodes are known for their high temp current leakage.  For the diode specified, it has  typical leakage at 125C and 10V of 250uA.  (room temp 100nA)  The datasheet specifies a absolute maximum of 20mA @ -100v and 100C, but realistically you probably never see more then 1mA. The biggest concern would be that the leakage would cause the motor to move.  I don't think that would happen with 1mA of current, but don't know.  This could be tested by powering the motor through a 12k resistor to 12V and see if any movement occurs.  

Really what should be done is try it on one board and see if power dissipation in the motor driver decreases and if the ~0.40 is worth it or not.  

The bigger issue is that the Motor driver IC has a absolute maximum rating of 3A vs the stall current of the motor being 14.6A (according to datasheet in github).  There is a fairly large voltage drop across the H-bridge since it is bipolar based design, so the actual stall current is likely lower, but still is  going to exceed the part ratings by a large amount turning the motor driver into a very expensive silicon fuse.  Fixing the diodes or adding better heatsinking isn't going to fix that issue.  

I design automotive electronics for a living, so I tend overdesign everything since returns from the field are very expensive.  If you are not getting large fallout in the field, it may not be worth changing anything.  

If you are still getting parts failing, The real way to improve the reliability is to go to a motor driver that has built in current limit, but that would involve a complete redesign to a new IC family.

blurfl, 2018-01-18:
The failures we've had were the H-bridge portions of the driver chips in the stall situation. It doesn't sound like the Schottky diodes would address this.
 I wonder whether the faster recovery diodes would make any noticeable difference in performance - our switching rate is 490 Hz at present.
 I've built a board using a different driver (TLE5206) which looks promising, but it uses a subtly different approach to control, and I haven't had the time to write a patch to drive it yet. I've got spare unpopulated boards if you're interested.