MaslowCNC/Electronics #47: Not enough Capacitance on 12V rail

dmouw25, 2018-01-13:
I should start this issue with a comment that I do not have a Maslow, so this is all theoretical.  

The 12V supply is a standard brick and i am assuming it is a 36" cable between power supply and Maslow PCB.   Signal travel along PCB or wires at ~ 6" per ns.  This means when additional current is required from power supply, it takes 12nS for power supply to respond (6 nanoseconds to get to power supply, then 6 nanosecond to get back).   

The basic capacitor equation is I=C DV/DT.   Assuming a stall current of 2amp , capacitance of 0.2uF and time of 12nS.  Rearranging to solve for Voltage, DV = I*DT/C and inputting values, we get DV =2*12nS/0.2uF.   DV = 120mV.  

Personally I think 120mV of ringing on the 12V line is unacceptable and would recommend adding  1uF capacitor to 12V reducing the ringing to 20mV.   This noise may be coupled into the encoder or USB traces causing issues with those signals.  

I would also add a 1uF to the 5V line, but that is more paranoia based on best practice.    I would recommend the 445-5146-1-ND from digikey and add to both 12V line and 5V line.  You can get 1uF capacitors in 0603 package, but we have found at work that 0603 can sometimes be damage by ESD strikes, so 0805 or greater should be used.  The .1uF capacitors could be reduced to 0402, but understand that some people find that hard to solder.  A 4.7uF or greater electrolytic capacitor could also be used, but it takes up more board space and is more expensive.

blurfl, 2018-01-13:
If I understand correctly, one should be able to see a ringing on the 12V plane in sync (at the same frequency as) with the PWM signal driving one of the motors at a moderately high load. One could look for that same frequency on the 5V line and encoder lines to check for crosstalk. 
 To date we haven't had a lot of issues that seemed cross-talk related, but conservative engineering is not a bad idea. It wouldn't be hard to add a couple caps to the design - near the point where the 12V and 5V line come onto the board, yes?

dmouw25, 2018-01-13:
That seems correct.  The ringing will repeat at the same rate as PWM, but the frequency of the ringing will be dependant on actual length of 12V harness.   It may also be prudent to add pads for a large electrolytic near 12V input, but skip by default.  a 6mm x 6mm such as Digikey part # 493-2190-1-ND would be perfect.

BarbourSmith, 2018-01-13:
I think adding 1uF capacitors to both the 12v and 5v lines is a good idea. @blurfl is right that we haven't seen much in the way of issues here, but capacitors are cheap and adding them is easy. 👍 👍

blurfl, 2018-01-14:
I'm going to hold the PR for this back until PR #48 is merged; this uses the schematic and board files from that branch. Don;t know how else to do this, trying to avoid a conflict.

BarbourSmith, 2018-01-15:
Smart!

blurfl, 2018-01-18:
Ok to close this?

BarbourSmith, 2018-01-18:
I vote to close

closed 2018-01-18 by blurfl