MaslowCNC/Electronics #50: Add a fuse

matmaxgeds, 2018-04-03:
It would be great to explore a fuse to prevent the motor drivers burning out when e.g. the chain wraps and the sprockets get stuck as suggested here: https://forums.maslowcnc.com/t/a-jam-burnt-the-motor-driver-board/3106/24

matmaxgeds, 2018-04-03:
If we made up some with the connectors, that would also be a good retrofit option for those who already have boards - bonus points if it is resettable

blurfl, 2018-04-03:
It will probably require s PCB for the male connector - it's only available as a [board-mounted device](https://www.digikey.com/product-detail/en/jst-sales-america-inc/B6B-XH-A(LF)(SN)/455-2271-ND/1000381)

[image]

The [mating connector](https://www.digikey.com/product-detail/en/jst-sales-america-inc/XHP-6/455-2218-ND/1651016)
[image]

and [pre-crimped pins](https://www.digikey.com/product-detail/en/jst-sales-america-inc/ASXHSXH22K305/455-3241-ND/6684932) for that connector
[image]

The board would make a good place for the fuse holder, though.

baverhey, 2018-07-20:
I was planning to add a fuse what would be the amperage needed and response time/curve?

BarbourSmith, 2018-07-20:
I can't give you an exact correct answer, but my instinct says something that would blow after about 250ms at 3 amps is probably a good starting point. Feedback on if that is a good number would be awesome.