MaslowCNC/Electronics #53: I need a z axis cable

Darkir1, 2019-01-02:
I don't have a z axis cable and I need to get one. what is the contact email?

BarbourSmith, 2019-01-03:
Unfortunately October 2018 was the last time that I had spare parts. I recommend contacting [MakerMadeCNC](https://forums.maslowcnc.com/u/makermadecnc/summary) through the forums to see if they can set you up with one. I'm sorry I can't send you one myself!

Darkir1, 2019-01-05:
I am currently putting together the cable I  need from online stock suppliers. I can start assembling packages for cables only anyone wants to buy one.

BarbourSmith, 2019-01-05:
I think that would be fantastic. Hopefully nobody else is missing one, but it could be useful for those who are assembling their own kits from parts. Listing the cable in [The Market](http://maslowcommunitygarden.org/?market=true) could be a good way to let people know it is available.

darigovresearch, 2021-04-17:
@Darkir1 Have you managed to solve this issue for yourself or put up a listing on the market?