MaslowCNC/Electronics #55: I got a micro usb cord....its supposed to be a printer style cord end?

ezgo123, 2020-11-14:
I ordered the maslow kit and im trying to connect everything only to find out i recieved a usb to micro usb cord...that doesnt connect to the board. I need a usb to printer style connector.

BarbourSmith, 2020-11-14:
That's wild. Who did you order the kit from?

ezgo123, 2020-11-14:
I ordered it through Amazon

GeroBH, 2020-11-15:
The Arduino DUE uses micro usb. Check your order if it was for an Arduino MEGA, then the cable is wrong. If your order was for the Arduino DUE you got the right cable but the wrong microcontroler.

BarbourSmith, 2020-11-15:
Ahhh @GeroBH  is correct. That sounds like the right USB cable. It could also be that you are following the wrong instructions. MakerMade CNC sells the kit you got so it might be worth checking to see if they have different assembly instructions

GeroBH, 2020-11-15:
MakerMade still sells a kit with the Mega and a newer kit with the Due, so the question is what kit did you order.
[image]