MaslowCNC/Electronics #57: [FEATURE REQUEST] Add Kitspace one click pricing

darigovresearch, 2021-04-17:
[Kitspace](https://kitspace.org/) is a project that really helps with building open source hardware as it allows you to (when properly implemented) have an almost one click pricing for the PCB and electronics components. If setup by the maintainers it would really decrease uncertainty if the components chosen are the ones that work reliably.

We haven't had to implement it yet but it seems relatively straightforward to implement and lets users have self-autonomy to replicate the work, build upon it and put it in the hands of users.

Examples of very good implementations of this are the [Open book](https://kitspace.org/boards/github.com/joeycastillo/the-open-book/open-book/) and [ebook wing](https://kitspace.org/boards/github.com/joeycastillo/the-open-book/ebook-wing/) projects by @joeycastillo

https://kitspace.org/1-click-bom/
https://github.com/kitspace/1clickBOM#readme

BarbourSmith, 2021-04-19:
I fully support this. Does the kit designer get a cut of the profits from the orders? If so I am absolutely OK with you putting it up there and keeping any money made from it.

darigovresearch, 2021-04-24:
@BarbourSmith I believe the cost only covers the price of the PCB and components but am sure a support button or profit sharing feature may be implementable. @kasbah could you please advise?

I am working on a pull request to add the project to kitspace but it requires gerbers to be exported so #59 needs to be resolved before it can be completed

kasbah, 2021-04-24:
Kitspace doesn't currently take any money except for the few kits I put together personally that are on shop.kitspace.org. The other orders all go directly to 3rd parties. The PCB companies listed are sponsors through [our Opencollective](https://opencollective.com/kitspace).

darigovresearch, 2021-05-01:
Have added a pull request #62 that would resolve the issue. @BarbourSmith I believe this must be merged before @kasbah can add the project to the kitspace website

BarbourSmith, 2021-05-03:
Great! Let's make it happen. Thanks for the pull request 👍

PR #62 (2021-05-03): Adds kitspace content to repo
Would resolve #57 if merged
closed 2021-05-03 by BarbourSmith