MaslowCNC/Electronics #59: [FEATURE REQUEST] Export Gerber files

darigovresearch, 2021-04-17:
Kitspace requires a folder of Gerbers so that they can be easily ordered from the various PCB suppliers they are linked to in order to get a price. It would also help for people who just want to order boards but don't have access to EAGLE

Could you export the Gerbers into a folder called `Gerbers` in the `PowerDistribution` folder?

We would do it but we don't have access to EAGLE

This would help with respect to issue #57 if you think it is worth us making a pull request

darigovresearch, 2021-04-17:
@kasbah can you confirm that Gerbers are required for Kitspace and the one click BOM to work?

kasbah, 2021-04-17:
At the moment, yes. We have been meaning to implement auto-plotting of Gerbers for Eagle for ages now. It's implemented for KiCad but not for Eagle.

BarbourSmith, 2021-04-19:
Unfortunately I also don't have access to Eagle anymore. Once Autodesk bought it it won't run on my computer anymore 😞

darigovresearch, 2021-04-24:
@BarbourSmith no worries, maybe someone from the community has access to it and could oblige. @kasbah or @midopple do you have access to Eagle to just make a pull request with exported gerbers in a suitable folder in [Power Distribution folder](https://github.com/MaslowCNC/Electronics/tree/master/PowerDistribution)?

May be worth trying to port the Eagle file into [KiCad](https://www.kicad.org/) to avoid the eagle issue for other users in the community

midopple, 2021-04-26:
Hi,

I have exported the Gerberfiles, see the Pull #61

darigovresearch, 2021-05-01:
@midopple thanks for resolving this issue!

closed 2021-05-01 by darigovresearch