MaslowCNC/Electronics #60: [FEATURE REQUEST] KiCad version of PCB

darigovresearch, 2021-04-24:
We may get more contributions/improvements to the system if it is in an open source PCB format like KiCad as it would reduce the barrier to entry for new users. Was wondering if the maintainers or someone from the community would be interested in trying this out.

https://www.kicad.org/

darigovresearch, 2021-04-24:
@midopple any chance you could do what you done to resolve https://github.com/drspangle/infidel-sensor/issues/24 here?

midopple, 2021-04-26:
Hi, 
i have added the Kicad Files #61
Please check it, at my first check i see no errors

[image]

darigovresearch, 2021-05-01:
Hi, @midopple thanks for being kind enough to tackle this issue. We're still learning KiCad but will raise any issues if we see any pop up so will close this as you managed to generate the files which is a great start!

closed 2021-05-01 by darigovresearch