MaslowCNC/Electronics #63: [BUG] Missing outline on the .mil file for the Gerbers

darigovresearch, 2021-05-03:
See https://github.com/MaslowCNC/Electronics/pull/62#issuecomment-830635273 for more details

Jamtek, 2021-08-01:
I noticed two large rectangular shapes outside the board outline. 
[image]

Removed the outside rectangles and regenerated the Gerber and drill files in KiCad. They now appear in the TraceSpace view portal as well as OSH Park.
![2021-08-01_18-31](https://user-images.githubusercontent.com/35613772/127787284-127eb73a-71bf-4d7b-9e
[image]
af-e5285739b7ce.png)

BarbourSmith, 2021-08-02:
I think those rectangles are the copper pours which create the ground plane and the power plane so they should be there.

Jamtek, 2021-08-04:
Ok, rookie mistake on my part. So maybe the actual issue was that the copper pours did not surround the entire board. Red Top pour is not surrounding the entire board outline. When I enlarged it to encompass the entire board and hit shortcut B to fill all zones, I was able to generate the Gerber files Edge_Cuts or board outline.
[image]

BarbourSmith, 2021-08-04:
I vaguely remember there being a reason for it not going all the way to the edge, but I can't remember what it would be. Is it OK to close this issue?